avrdudes/avarice #63: [patch #29] Range stepping for GDB 7.7+

avrs-admin, 2022-01-26:
m-j-h <None>
2014-03-21 17:07:21.141000

GDB 7.7 introduced range stepping, in combination with the vCont remote protocol extension. This patch series includes support for these features and implements them for JTAG ICE mkII devices only, as we only have documentation for those.
Debugging, especially for single steps and function calls is greatly increased by this, comparable to AVRStudio on Windows.

[0004-Obsoletes-Implement-range-stepping-GDB-7.7.patch](https://sourceforge.net/p/avarice/patches/_discuss/thread/a45eda86/8f94/attachment/0004-Obsoletes-Implement-range-stepping-GDB-7.7.patch)
[0005-Disable-interrupts-while-stepping.patch](https://sourceforge.net/p/avarice/patches/_discuss/thread/a45eda86/8d5d/attachment/0005-Disable-interrupts-while-stepping.patch)

This issue was migrated from https://sourceforge.net/p/avarice/patches/29/

avrs-admin, 2022-01-26:
cdealti
2015-05-30 20:19:52.202000

This patch obsoletes 0004-Implement-range-stepping-GDB-7.7.patch.
Interrupts are ignored while range stepping if the --ignore-intr option is specified.
Otherwise, like in the original patch, gdb will break in the interrupt vector table.

avrs-admin, 2022-01-26:
cdealti
2015-08-17 14:02:15.999000

This patch disables interrupts while stepping, like in AVR Studio.

Compared to using the --ignore-intr option,
this improves debugging speed because the target will not
break at every interrupt while running to end address or singly
stepping.
Note: EXPERIMENTAL. There are cases where restoring the interrupt
status is wrong. Read the detailed comments in the source code.

cdealti, 2022-10-23:
Adding the links to the original patches authored by m-j-h:
[0001-Implement-setBreakOnChangeOfFlow.patch](https://sourceforge.net/p/avarice/patches/29/attachment/0001-Implement-setBreakOnChangeOfFlow.patch)
[0002-Implement-run-to-address-method-for-mkII.patch](https://sourceforge.net/p/avarice/patches/29/attachment/0002-Implement-run-to-address-method-for-mkII.patch)
[0003-Implement-flag-if-device-supports-range-stepping.patch](https://sourceforge.net/p/avarice/patches/29/attachment/0003-Implement-flag-if-device-supports-range-stepping.patch)
[0004-Implement-range-stepping-GDB-7.7.patch](https://sourceforge.net/p/avarice/patches/29/attachment/0004-Implement-range-stepping-GDB-7.7.patch)