avrdudes/avarice #85: [feature-request #11] Support UPDI programming

avrs-admin, 2022-01-27:
maxgerhardt <None>
2021-02-12 17:27:23.497000

The fork https://github.com/Florin-Popescu/avarice contains some bleeding-edge expansions to avarice regarding programming / debugging microcontrollers with the UPDI interface. I think that's very cool and would eventually allow debugging of e.g. the Arduino Nano Every board with no extra debug-probe and those features should definitely be merged back. Maybe even a collaboration is possible?

This issue was migrated from https://sourceforge.net/p/avarice/feature-requests/11/

maxgerhardt, 2023-01-14:
I still think this is really important. I've also opened https://github.com/Florin-Popescu/avarice/issues/9 in regards to this.

mcuee, 2023-03-11:
PR has been submitted.
* https://github.com/avrdudes/avarice/pull/101