avrdudes/avarice #97: Add mEDBG support

maxgerhardt, 2023-01-17:
I was quite baffled when I came accross http://www4.cs.fau.de/Lehre/WS18/V_GSPIC/SPiCboard/zuhause_linux.shtml referencing an `avarice -5` option that I have not seen yet, claiming it is mEDBG probe support. And indeed, their fork from 4 years ago apparently looks like it it has mEDBG supports:

https://gitlab.cs.fau.de/i4/spic/tools/AVaRICE/-/blob/master/avarice/src/main.cc#L173-174

Any improvements and additions of this version should be backported to here.

maxgerhardt, 2023-01-19:
To my suprise, the current version of AVaRICE can connect to an mEDBG option via the regular `--edbg` command just fine.. 

[image]

I need to inspect the exact diff on what they changed there or saw necessary to change.

felias-fogg, 2025-12-19:
Connection works, but there are some hiccups in breakpoint handling. I have a small example program, where setting two breakpoints before starting leads to a loss of connection to the OCD when execution starts. Very strange.