avrdudes/avarice #98: Document needed udev rules for USB HID/CDC devices

maxgerhardt, 2023-01-17:
It may seem obvious to a long-time Linux user, but `avarice` for me failed to retrieve the USB product string of the enumerated devices in `openhid()` (`walk->product_string` being a `nullptr`), but that's just when udev rules aren't installed so that the regular user has access to the e.g. CMSIS-DAP based HID devices. The websit linked in #97 mentions this, but doesn't give a general enough udev file to directly use. We should come up with one.

maxgerhardt, 2023-01-17:
Indeed it was udev rules. Initially
[image]

now
[image]

after adding `/etc/udev/rules.d/99-atmel-medbg-cmsis.rules` as 
```
# mEDBG CMSIS-DAP
SUBSYSTEM=="tty", ENV{ID_REVISION}=="03eb", ENV{ID_MODEL_ID}=="2145", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_CANDIDATE}="0"
SUBSYSTEM=="usb", ATTR{idVendor}=="03eb", ATTR{idProduct}=="2145", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1"
```
and doing
```
sudo udevadm control --reload-rules
sudo udevadm trigger
```
and replugging the device.

dl8dtl, 2023-01-18:
The question is *where* this might best be documented.

The `doc` subdirectory contains a bit of documentation, but nothing like installation instructions.

I think it makes sense we add a toplevel `README.md`, now that the project is on Github, and document things like basic installation and setup there.

Opinions?

dl8dtl, 2023-01-18:
Btw., the usual (Linux) policy would not use mode `0666`, but rather `0660`, and assign to group `plugdev`.

Legitimate users are then supposed to be members of `plugdev` in order to attach that kind of devices.