avrdudes/avarice #99: Add proper README.md

maxgerhardt, 2023-01-17:
This project repository lacks a proper representative README.md that would give users much needed insight on what this repo is without vising other websites.

Florin-Popescu, 2023-02-08:
Hi!

Partly related, the [README](https://github.com/Florin-Popescu/avarice/blob/master/README.md) in my fork documents some steps on how to extend the project in the future. Would that be good to include in the readme here or in some other file? I would like to delete my fork after everything is upstreamed here, and I think I've gathered some good information that should be saved somewhere.

dl8dtl, 2023-02-08:
I think this would be more suited for an issue tracker or discussion item type.