avrdudes/avarice #100: Writing Flash from GDB fails

maxgerhardt, 2023-01-19:
When using an avr-gdb that was not compiled with XML memory parsing enabled, avarice seems to refuse to write data to flash if the incoming data is not in a page-size.

I tried both avr-gdb bundled in the 7.3.0 Arduino toolochain as well as the one bundled in the 12.1.0 toolchain found at https://blog.zakkemble.net/avr-gcc-builds/.

When a `load` is executed, the following happens.

GDB
```
> avr-gdb.exe -batch -ex "target extended-remote :3232" -ex "load" firmware.elf
0x00000000 in __vectors ()
warning: Can not parse XML memory map; XML support was disabled at compile time
Loading section .text, size 0xa14 lma 0x0
Remote connection closed
```
AVaRICE

```
>avarice.exe --edbg --debugwire --debug :3232
[...]
->GDB: 0000b7fbdf92effebed7f7f3dfcd7f6ff7dd0100808fdf8f08007d6ee2ff00c400ff0800000000
GDB: <qL1160000000000000000>
->GDB:
GDB: <qSymbol::>
->GDB:
GDB: <qXfer:memory-map:read::0,18a>
->GDB: l<memory-map>
  <memory type="ram" start="0x800000" length="0x20000" />
  <memory type="flash" start="0" length="0x8000">
     <property name="blocksize">0x80</property>
  </memory>
</memory-map>

GDB: <X0,0:>
->GDB:
GDB: <M0,20:0c9466000c9483000c9483000c9483000c9483000c9483000c9483000c948300>

GDB: Write 32 bytes to 0x0
jtagWrite jtagWrite(): numByte does not match page size
```

While I'm trying to working on recompiling avr-gdb that *does* have XML memory support enabled, would make sense to make avarice more "forgiving", in the sense that if a write to page is requested with data that is noth page-length long, to first read the current page, overwrite the part with the wanted data while keeping the other old data?

maxgerhardt, 2023-01-20:
Nice, I managed to recompiled avr-gdb with the `--with-expat` configuration option, and the resulting AVR-GDB binary then correctly sends the data to AVaARICE with the right block size, flashing it :)

dl8dtl, 2023-01-20:
The more complicated memory layouts used by MCUs basically require that you use the XML stuff to describe them.
Consequently, without enabling expat in GDB, there's no chance to upload memory contents.

maxgerhardt, 2023-01-20:
I see, then it makes more sense to require XML support for avr-gdb when flashing is supposed to be done via AVaRICE instead of patching AVaRICE.

closed 2023-01-20 by maxgerhardt